plottertools/vpype-gcode #2: Allow storing the "version"/settings somewhere as a default

theomega, 2021-01-25:
I have configured quite complex settings for my pen plotter, but I have no chance of saving them as defaults. Not sure what would fit the design of this software well. Hardcoding them in the actual source code seems not a great idea because it will be lost with an update. I also think my settings are very special, so they should not become part of this repo

theomega, 2021-01-25:
Just for reference, these are my current defaults:
```
vpype read [...] gwrite --preblock '(Start Block)\n' --move 'G00 Z5\nG00 X%.4f Y%.4f\nM3 S1000\nG4 P0.3\nG01 Z1\n' --line 'G01 X%.4f Y%.4f Z1\n' --header 'M3\nG21\n\n\n' --postblock 'G00 Z 5.0000\nM5 S0\nG4 P0.5\n(End Block)\n\n' --prelayer '(Start Layer)\n' --postlayer '(End Layer)\n\n\n' --scale 0.2645 [filename]
```

tatarize, 2021-01-26:
Yeah, not sure where the temp stuff is stored. I might have to ask @abey79 where he put those other settings. Since this was more intended to get him to include such a writing engine inside vpype. But, I think he has some settings for the hpgl that would make a useful thing.

abey79, 2021-01-26:
vpype can load .toml config file with `vpype -c config.toml ...`. It will automatically load `~/.vpype.toml` too, and whatever config file is included in the distribution. TOML are essentially like JSON. They load as a big nested dict/list/etc. In the plug-in, you can access the config this way:

```python
import vpype as vp

print(vp.CONFIG_MANAGER.config)  # big nested dict with all config
```

From there, just specify the format you like, e.g:

```toml
[gwrite.config1]
prelayer = "(Start Layer)\n"
postlayer = "(End Layer)\n\n\n"
scale = 0.2645
header =  "M3\nG21\n\n\n"
# ...

[gwrite.config2]
prelayer = "..."
# ...

# ...
```

The user can follow that format and put the config they like in `~/.vpype.toml` and use:

```
vpype read input.svg [...] gwrite --config config1 output.gcode
```

The plugin would look like:

```python

if "gwrite" not in vp.CONFIG_MANAGER.config or config_name not in vp.CONFIG_MANAGER.config["gwrite"]:
    raise Exception("config not found")

config = vp.CONFIG_MANAGER.config["gwrite"][config_name]

pre_layer = config.get("prelayer", "")
scale = config.get("scale", 1.0)
# ...
```

tatarize, 2021-01-26:
That seems perfect. I'll move the couple user pref configs in there and do that. Likely be a day or so, I have a big buggy project ready to get some feedback shortly. But, this is pretty solid.

theomega, 2021-01-26:
What do you guys think about this:

https://github.com/theomega/vpype-gcode/pull/2

The README would need to be updated to explain where to put the config file. I moved the config from the source 1:1 into the example.

abey79, 2021-01-26:
@tatarize if you want to package a config file with your plug-in, just add it in the manifest and call this at the top of the file:

```python
from pathlib import Path

vp.CONFIG_MANAGER.load_config_file(str(Path(__file__).parent / "bundled_configs.toml"))
```

PR #9 (2021-01-28): Move plotter configuration to configuration file
Removes the possibility to provide the configuration strings
via the command line and moves the config over to the vpype
configuration.

This is still WIP, the README would need to be updated if we
decide to merge this.

Closes #2
closed 2021-01-28 by tatarize