plottertools/vpype-gcode #8: Use `.format()` and keyword argument for string formatting and remove `relative` parameter

abey79, 2021-01-26:
For flexibility and easy of extension, I suggest using the following pattern for string formatting:

```python
f.write(move.format(dx=dx, dy=dy, x=x, y=y)
```

Using this approach, all of the following configuration actually work as expected (not that they would be correct g-code):
```
move = "G00 {x:.4f} {y:.4f}\n"
move = "G00 {y:.4f} {x:.4f}\n"
move = "G00 {dx:.4f} {dy:.4f}\n"
move = "G00 {dy:.4f} {dx:.4f}\n"
```

Benefit:
* the `relative` parameter can be removed
* more flexibility of the user
* more flexibility to add many other variable that may be useful to the user (e.g. line_number, segment_number, layer_id, etc.,)

Again, since this changes the configuration file format (user must use the {} notation), this change is best done early.

tatarize, 2021-01-28:
This is certainly a solid request. I'll see if it ports across rather trivially, though I would assume it would.

tatarize, 2021-01-28:
Completed in 0.4.0

closed 2021-01-28 by tatarize