plottertools/vpype-gcode #24: Possible to look back/ahead?

pomeroyb, 2022-06-18:
Hello! I'm trying to use vpype-gcode with a controller that is expecting commands to tell it to travel to a point and draw a line. Essentially the "point" function is used as a G0 command, and the "line" function is used as a G1 command. The point command is easily solved with "segment_first", but the line command needs to look like the following:

```python
line((start_x, start_y), (end_x, end_y), speed = 4)
```

Right now my custom profile looks like this:

```toml
[gwrite.pewpew]
document_start = ""
segment_first = "point(({x:.4f},{y:.4f}), delay = 0.25, burn = 0)\n"
segment = "line(({x:.4f},{y:.4f}),({x:.4f},{y:.4f}), speed = 4)\n"
document_end = ""
unit = "in"
invert_y = true
info= "This gcode profile is correctly inverted across the y-axis"
```

The problem is that the segment cannot (as far as I can tell) refer to the coordinates of the previous segment, so I end up making a zero length line at the coordinates of the current segment. Ideally my toml would have the following:

```toml
segment = "line(({prev_x:.4f},{prev_y:.4f}),({x:.4f},{y:.4f}), speed = 4)\n"
```

Is there a way to do this in vpype-gcode or is this outside of the capabilities of the project currently?

tatarize, 2022-09-22:
It's outside the current capabilities. Though it could probably be added. There's some entries for `dx, dy` so if you could use that rather than the actual previous value you'd be in luck.

abey79, 2022-09-23:
Well we have `last_x`/`last_y` lying around for the computation of `dx`/`dy`. Wouldn't it be trivial to just add them to the template variables? Am I missing something?

tatarize, 2022-09-23:
Super trivial. It's an oversight it already doesn't work.