plottertools/vpype-gcode #25: Invert Y axis for the whole document

brianlow, 2022-07-09:
I'd like to draw SVGs in the same orientation they will appear on the plot. 

My plotter's origin is bottom left.

I am finding `invert_y: true` flips a shape along the shape's center. 

Is there an option to flip along the document's center?

[image]

<details>
  <summary>Steps to Reproduce</summary>

  ```
    vpype --config "${PWD}/vpype.toml" read t.svg gwrite --profile default t.gcode
  ```
  vpype.toml:
  ```toml
[gwrite.default]

invert_y = true

segment = "x X{x:.4f}, y Y{y:.4f}\n"

  ```
  t.svg:
  ```svg
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="100%" height="100%" viewBox="0 0 783 651" version="1.1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" xml:space="preserve" xmlns:serif="http://www.serif.com/" style="fill-rule:evenodd;clip-rule:evenodd;stroke-linecap:round;stroke-linejoin:round;stroke-miterlimit:1.5;">
    <path d="M11.915,624.943L23.83,650.079L0,650.079L11.915,624.943Z" style="fill:none;stroke:black;stroke-width:1px;"/>
</svg>

  ```

  t.gcode:
```gcode
x X3.1525, y Y172.0001
x X6.3050, y Y165.3495
x X0.0000, y Y165.3495
x X3.1525, y Y172.0001
```

^^^ expected the Y coordinates to be close to 0

</details>

tatarize, 2022-07-09:
I believe there is one but it would need to be in vpype actual, since I believe the document itself needs to be flipped. Though if you know the size of the document you could offset by the document height. But, vpype-gcode isn't given the document size information. @abey79 how/where is that document info stored.

This is entirely correct, the coordinate system for gcode is typically origin in bottom-left rather than origin in upper-right. IS there a flip within document code that does this correctly? Because vpype gcode doesn't know the actual document size to do that math (or I dunno how to access it exactly.

brianlow, 2022-07-09:
Thanks! That pointed me in the right direction:
```
vpype  ...
   read ...
   scale --origin 0 86mm -- 1 -1     # 86mm is the document center
   gwrite ...
```
I know the document size before hand (it is my plotter's printable area):

Closing because solved. 

Would be interested in a solution without hardcoded center if anyone finds one. The `stat` command knows the document size so might be something there:
```
========= Stats ========= 
Current page size: (783.0, 651.0)
Totals:
  Global properties:
    vp_page_size: (783.0, 651.0)
```

closed 2022-07-09 by brianlow