plottertools/vpype-gcode #32: Center geometry for gcode

ahmed2007, 2023-01-19:
Dear all,
I tried to center the geometry for HPGL and it is working well, using the option "--center" or "layout".

However i can't have the same option with gcode. So can i have my geometry center with gcode ? is there any option for gwrite ? or am i missing something in the difinition of my profile in the .tom file. 
I left here the profile i created:
```
[gwrite.gcodemmahmed]
document_start = "G17\nG71\nG90\n"
segment_first = "SC_TOOL[0]\nG0 X{x:.4f} Y{y:.4f}\n"
segment = "SC_TOOL[1]\nG1 X{x:.4f} Y{y:.4f}\n"
document_end = "G0 X0 Y0\nM2\n"
unit = "mm"
vertical_flip = true
info= "This gcode profile is correctly inverted across the y-axis"
offset_x = 0
offset_y = 0
```
Thank you guys!

emmano3h, 2023-01-19:
Have you tried vertical_flip = false ?

tatarize, 2023-01-19:
invert_x: Flip all points right-to-left without changing the position within the document.
invert_y: Flip all points top-to-bottom without changing the position within the document.

This was previously noted. The vertical flip makes all Y into -Y. When what you actually want is the document unmoved and flipped in place. That's what `invert_y = true` should give you.

ahmed2007, 2023-01-19:
Thank you tartize, actually i don't want to flip my document, what  i want is to have my document centered in my paper, that is to say the document moved to the center. As i say in my issue description, i got the document centred with vpype using the option "--center" to have HPGL, is there a way to have the same thing with gcode ?
Thanks a lot !

tatarize, 2023-01-21:
So you just want to translate the design? Seems like there's a bunch of commands in vpype for that. At least if you know the size... @abey79 is there a --center that exists outside the `write` routine or does he need to check the stats and do it manually?

tatarize, 2023-01-21:
Apparently something like `%vp_page_width.h/2%` will let you do that sort of stuff and evaluate it.

I think if you do `translate %vp_page_width.w/2%` or something or other you can center it. There's currently not a thing to do that built into vpype-gcode and seems like that stuff should be doable in vpype proper, especially with the evaluation stuff.